prusa3d/PrusaSlicer #11353: PrusaSlicer crash during slicing (problem with Win 11 or hardware ?)

lwiart, 2023-09-23:
### Description of the bug

Hello,

I installed PrusaSlicer 2.6.1+win64 on my Windows 11 desktop PC.
I'm trying to slice a model from Printables (https://www.printables.com/model/563912-safe-from-the-rain) but it crashes when I press "Slice Now".

I configured a new printer size 256x256x256 (for my Bambu Lab X1C) and 0.2 nozzle size.
Nothing fancy, all other params default.

I'm wondering if it is a problem with my PC hardware since it also crashes with Cura and Bambu Studio...
hardware :
- Core i9-14900K
- Asus ROG Maxiums Z790 Hero
- Corsair vengeance 2x32 6600 MHz
- Asus 4090 GPU

Thanks for your help

### Project file & How to reproduce

- Open .3mf file in Prusaslicer
- Press "Slice now"
- PrusaSlicer crashes (closes) after a couple of seconds

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1+win64

### Operating system

Windows 11  10.0.22621 Build 22621

### Printer model

Bambu Lab X1C

Jan-Soustruznik, 2023-09-25:
Hello, @lwiart, could you share your project file(3MF), please.  Thank you, have a nice day

lwiart, 2023-09-25:
Hey @Jan-Soustruznik ,

Thank you for getting back to me.

I've got new findings.
It doesn't seem to be a PrusaSlicer issue, but rather an issue with my CPU (i9-13900K) : 
- When I slice right away with PrusaSlicer -> they crash after a few seconds before finishing the slicing.
- But when I set the CPU affinity of the process (prusa-slicer.exe), and choose for example CPU0 to CPU3 -> no problem with slicing!

So I don't think it's a bug with PrusaSlicer, but a problem with my CPU, specifically some core(s), or with multi-threading...

lwiart, 2023-09-28:
Hey @Jan-Soustruznik ,

I got to the bottom of the problem.
It wasn't related to PrusaSlicer but to a problem with my ASUS motherboard (actually many recent ASUS motherboard models) and my i9-13900K.

A parameter in the Bios has to be changed : SVID Behavior (change from ""auto" to "Intel's Fail Safe").

lukasmatena, 2023-10-04:
@lwiart Thanks for getting back to us. From what I found online, it really appears not to be a problem in PrusaSlicer. I am closing the issue.

closed 2023-10-04 by lukasmatena