prusa3d/PrusaSlicer #11355: Does anyone have a solution to this annoying problem? Unfortunately I only have a Mac.

yasars, 2023-09-23:
### Description of the bug

no upload to connect ... 

### Project file & How to reproduce

Does anyone have a solution to this annoying problem? Unfortunately I only have a Mac.

[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

latest for mac

### Operating system

mac os 

### Printer model

mk4

bdaroz, 2023-09-24:
Also having a lot of problems getting stuff to upload to connect. Most often I get this
[image]

But when it does connect and start the upload, I get the same error in OP 100% of the time. I have to export and go to the web interface to upload.

yasars, 2023-09-24:
Yes over the website you can only upload to connect files ...not to the printer...

bdaroz, 2023-09-24:
Looks like a dupe of #11076

Jan-Soustruznik, 2023-09-26:
thank you for report Problems seems to be similar like https://github.com/prusa3d/PrusaSlicer/issues/11076

Problem is caused by connect service. We just display only the ERR message that we get from connect. We informed Connect team about this issue. 

I closed issue as duplicate. 

Have a nice day.

closed 2023-09-26 by Jan-Soustruznik

lukasmatena, 2023-11-11:
There is an update in https://github.com/prusa3d/PrusaSlicer/issues/11076, we would be grateful if you guys also had a look. Thanks.