prusa3d/PrusaSlicer #11358: Spiral Vase - Why mess with other settings? it could be so easy!

hcirnieh, 2023-09-23:
It's so easy I just need 1 picture to describe what I mean:
[image]

Ignore those **_while_** Spiral Vase is enabled (but please don't change the whole set of settings).

wuyoutech, 2023-10-24:
can't agree more,  just 'disable' or 'override' them. it could be useful to restore everything when closing Spiral Vase mode

lit017, 2023-10-25:
> can't agree more, just 'disable' or 'override' them. it could be useful to restore everything when closing Spiral Vase mode

Exactly! I want to use my main profile for vase mode by quickly flicking a switch. 
Especially since I change settings a lot this would be very helpful.

hcirnieh, 2023-11-21:
> I agree.
> Also if "spiral vase" could be added as part modifier, would be super helpful for us who save 1 project with multiple parts that print on different printers. Thanks

That would also make printing a compound of solid and vase mode parts easier.

closed 2026-09-29 by prusaslicer-bot[bot]