prusa3d/PrusaSlicer #11359: Sync settings between multiple devices

Gona2119, 2023-09-23:
**Is your feature request related to a problem? Please describe.**
Sometimes I just want to slice a model on my desktop PC and other times I want to slice any model with my notebook, sometimes the settings and profiles differ a little bit from one PC to other, so would be awesome to just can sync those files.

**Describe the solution you'd like**
I was thinking about the possibility to tell PS where to store and load the profiles and setting files so then I just can save the files with Google Drive for example. Then when I open PS in other pc with Google Drive installed, the first time I start PS, it would search for those files in my virtual drive created by GDrive

**Describe how it would work**
Was explained in the past item

**Describe alternatives you've considered**

Other solution could be just to implement any account or cloud storage to only save those files, or using any "Link Code" to import and save entire settings. The first example of this second method could be the "crosshair code" of Counter Strike Global Offensive kinda

**Additional context**
Nothing to add

mix579, 2023-09-24:
I would certainly appreciate a cloud-based sync solution. But you can get close to it even today with symlinks.

* Go to Prusaslicer's Help menu and select Show Configuration Folder
* Go up one level and note the location/path name of the folder. It's usually /Users/username/Library/Application Support/PrusaSlicer (alphas and betas have different names). On Windows, it would look like C:/Users/username/AppData/Roaming/PrusaSlicer or similar.
* Create a backup copy, just in case.
* Copy the configuration folder to your network drive and delete the original. Note the new pathname, in my case /Users/username/OneDrive/3D Printing/Software/PrusaSlicer/PrusaSlicer
* Open a terminal window.
* Create a symlink on each computer on which you want to sync it:
ln -s /Users/username/OneDrive/3D Printing/Software/PrusaSlicer/PrusaSlicer /Users/username/Library/Application Support/PrusaSlicer

What all this does is create a "master copy" of the config folder on a network drive (OneDrive/Gooogle drive/Dropbox handle the syncing of that drive across machines in the background). The symlink creates a pointer to that folder in the location PrusaSlicer looks for the config folder.

Another option is to use the -datadir command line option to redirect Prusaslicer to look for a network drive.

Gona2119, 2023-09-24:
Yeah, I knew that was possible. Ty for the tutorial because in other places it's quite confusing. Despite of that it would be pretty nice that the devs implement this into PS

closed 2023-09-24 by Gona2119