prusa3d/PrusaSlicer #11361: why does PrusaSlicer strictly depend on version 7.6.2 of opencascade occt?

evils, 2023-09-24:
hi, i maintain a package in Nixpkgs which shares a dependency with PrusaSlicer: opencascade's `OCCT`

due to [PrusaSlicer strictly depending on version 7.6.2 of OCCT](https://github.com/prusa3d/PrusaSlicer/blob/1a4a31e92992090948e64984479388d47797142e/src/occt_wrapper/CMakeLists.txt#L22),
[Nixpkgs has been holding back on updating OCCT for over a year](https://github.com/NixOS/nixpkgs/pull/149796#issuecomment-1253927727)

i was assuming the strict dependency would be resolved with newer releases of OCCT and PrusaSlicer

removing the version part of the dependency discovery and using OCCT 7.7.2 seems to have no ill effect
and i cannot find an issue indicating what kind of problem to look for

@tamasmeszaros you seem to have set that version and fairly active around here,
can you shed some light on the matter?

bubnikv, 2023-09-24:
Compiling against a tested version is a good practice to avoid unwanted
surprises.
We certainly do want to upgrade versions of those dependencies at the start
of each major development cycle.

evils, 2023-09-24:
that sounds like the caution of someone who also had to deal with wxwidgets :)

updating on every major release seems like a good idea, but 2.0.0 was 4 years ago and it always was 1.x before then
would reassessing strict dependencies be possible on minor releases? (which you appear to do every ~6 months)

or be strict about what you use to compile your release binaries, but allow building with newer versions?
maybe with a warning when not using the specific version?

note that [FreeCAD isn't that strict](https://github.com/FreeCAD/FreeCAD/blob/97d84ec1842bc19f79e5c240510e5cf111ec2bc4/cMake/FindOCC.cmake#L39), nor seems [KiCad](https://gitlab.com/kicad/code/kicad/-/blob/master/CMakeLists.txt#L797)

for nixpkgs i see 3 solutions:
- leave OCCT out-of-date for every package (the current approach)
- repackage 7.6.2 just for PrusaSlicer (using [OCCT 7.6.3](https://github.com/Open-Cascade-SAS/OCCT/releases/tag/V7_6_3) would be preferred if you want to stick to their minor releases)
- override PrusaSlicer to accept the version we have, which would then likely be the latest

overriding the dependency is my current preferred approach, any words of caution / where to look for problems?
(all your tests pass and the GUI seems to work)

i should have probably done this at the start
ping @Moredread and @thorstenweber83, the listed maintainers of the Nix package

adelton, 2024-05-31:
For the record, in Fedora I just patched that `find_package` to search for 7.6.3:
https://src.fedoraproject.org/rpms/prusa-slicer/c/aed80025dcdddb59057de13269b364436037bfbb

Do you say that that you are able to go as high as 7.7.2, by simply omitting the version clause?

That would actually help us unblock rebase of opencascade (https://bugzilla.redhat.com/show_bug.cgi?id=2264941).

evils, 2024-06-01:
> Do you say that that you are able to go as high as 7.7.2, by simply omitting the version clause?

that is my recollection and interpretation of my message too

note that nixpkgs took the more conservative approach of packaging 7.6.2 separately and using that for prusa slicer
https://github.com/NixOS/nixpkgs/pull/309347

and skipped straight to occt 7.8.1, with which prusa slicer 2.7.4 does not seem to build

here's the patch i believe i used to build with 7.7.2
```
diff --git a/src/occt_wrapper/CMakeLists.txt b/src/occt_wrapper/CMakeLists.txt
index 8ab9fb4be..d8dd8e1d4 100644
--- a/src/occt_wrapper/CMakeLists.txt
+++ b/src/occt_wrapper/CMakeLists.txt
@@ -19,7 +19,7 @@ include(GenerateExportHeader)

 generate_export_header(OCCTWrapper)

-find_package(OpenCASCADE 7.6.2 REQUIRED)
+find_package(OpenCASCADE REQUIRED)

 set(OCCT_LIBS
     TKXDESTEP
```

closed 2026-09-29 by prusaslicer-bot[bot]