prusa3d/PrusaSlicer #11362: Slow transfer from Prusa slicer to all prusa printers

NeuralDEEPnet, 2023-09-24:
### Description of the bug

Upload and print is very slow, oftern error occurs . 

### Project file & How to reproduce

Send any file bigger than 5mb to print and it takes 15 min + 

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version: 2.6.1

### Operating system

+win64 Build: PrusaSlicer-2.6.1+win64-202309060711 Operating System: Windows System Architecture: 64 bit Windows Version: Windows 11 (build 22631), 64-bit edition Total RAM size [MB]: 51,449MB

### Printer model

mini+, XL, mk4

PaparazziN, 2023-10-21:
Form my understanding the network card used on the XL, mini+ and MK4 is quite slow (might be different models but they all seem to be quite slow). So the core issue is most likely not with PrusaSlicer and rather with the hardware which is a bit unfortunate. With the latest Alpha (2.7.0-Alpha1 at the time of writing) they have released some features that significantly reduce the size of G-codes, such as G2/G3 arc support and the binary G-code format. There is more info in the release notes.

With that said, I think it would be appropriate if you close this issue. If the Alpha release still is problematic with regards to this, I would open a separate issue with more specifics for that.

lukasmatena, 2023-10-23:
As @PaparazziN has said, this is not PrusaSlicer related issue. The new [5.1.0-alpha2 firmware](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0-alpha2) has added some remedies (G-code streaming in particular) and the new [PrusaSlicer 2.7.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-alpha1) also contains some related features (binary G-code, G2/G3 G-codes).

Closing.

closed 2023-10-23 by lukasmatena