prusa3d/PrusaSlicer #11364: Color change G-Code Color Hex Code for M117

dabohda, 2023-09-24:
I would like to display the color code for the next color change with my costum color change G-Code:

M117 Change Color to {extruder_colour[0]}
M600

But this allways display the fix color for the given extruder set in the settings. Is there any variable i can use in the gcode to display the hex color code?

Is there any other variable that returns the color hex code?

I have seen that in the generate Gcode there is what i want but commented:

;COLOR_CHANGE,T0,#000000
M117 Change Color to #FFFFFF
M600

Thx for any help.

JacobChrist, 2023-09-25:
I would be happy if Prusa Slicer supported adding the extruder number to the M600 command (maybe it already does) then if there is no MMU it could just display the "index" of the desired color.  This would allow testing multi color prints without an MMU.  If this feature was implemented Prusa would probably would sell a lot more MMU's if people can test out how multi color printing works on there parts.

JacobChrist, 2023-09-25:
This is related to #11265

rcmaniac25, 2023-10-25:
Literally making tickets of things I wanted and this was the next one on my list to write. It may be more this is an option for a printer, and then the printer itself needs the support. Otherwise a comment may be overwritten my the M600 text instructions... making the extra gcode kinda useless.

Harry-58, 2024-11-20:
I would also find it very helpful if the color for the COLOR_CHANGE, were available as a variable to use it in custom-G-code
And not just as a comment in the Gcode:
;COLOR_CHANGE,T0,#FFFFFFFF

closed 2026-09-29 by prusaslicer-bot[bot]