prusa3d/PrusaSlicer #11367: Arrange puts wipetower outside print area

mozgy, 2023-09-24:
### Description of the bug

Load two color benchy from printables, only objects fom 3mf,
click Arange
click Slice

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

Windows 10

### Printer model

Prusa XL 2H

lukasmatena, 2023-10-03:
Thanks. Can you please provide a 3MF?

mozgy, 2023-10-12:
Really ? **All info** is in the first line of my bugreport ..
https://www.printables.com/model/32815-two-colour-benchy

lukasmatena, 2023-10-13:
@mozgy Yes, really. We need a 3MF project saved exactly when that happens. I am struggling to reproduce the issue, it seems to somehow depend on configuration of the wipe tower, and the **All info** that you think you provided does not contain it. And the 3MF on Printables is for an MK3, not an XL.

The only way I managed to reproduce it was with extremely wide wipe tower brim, but it may be a separate issue.

Thanks.

mozgy, 2023-10-13:
Ok, noted, here are my steps -
start PrusaSlicer 2.6.1
choose XL 2T
click on that link, download 3mf
load 3mf, choose import 3d model only
click Slice Now
*burp*
[image]

mozgy, 2023-10-13:
Here's saved 3mf immediately after that picture - 
[bugreport.zip](https://github.com/prusa3d/PrusaSlicer/files/12894684/bugreport.zip)

lukasmatena, 2023-10-13:
Thanks, we will look into it.
SPE-1966

mr-ice, 2024-02-05:
I'm seeing this ~4 months later, any updates?  

The workflow @mozgy  reported exactly reproduces it.   However, this is not the ONLY workflow that reproduces it.  I get the same output putting any two models on, changing the extruder for one of them, and doing [A]rrange.

Moving the wipe tower (default settings here, the arrange seems to have only moved the wipe tower and not changed anything else about it -- see screenshot).  Moving it back (-Y) 0.2mm is enough to avoid the problem.

Arrange should take into account all the wipe tower settings and put it completely inside the print area.
[bugreport.zip](https://github.com/prusa3d/PrusaSlicer/files/14159159/bugreport.zip)

[image]

[image]

lukasmatena, 2024-02-05:
It will be fixed in the next release.

lukasmatena, 2024-02-09:
Fixed in [2.7.2-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.2-alpha2). Closing.

closed 2024-02-09 by lukasmatena