prusa3d/PrusaSlicer #11373: Bridging in an annular circle

RetireeJay46, 2023-09-25:
I have a part which requires bridging, but the area that needs bridging is an annular circle.  With standard bridging, the slicer always runs the bridging traces in a single direction, which makes some of the bridges very long.  I wish there were a way to break up the bridging into quadrants, or even have "radial" bridging.  The part actually does work, but some of the bridges are a little sloppy and the slicer (2.6.1) complains.
[image]

themzlab, 2024-01-12:
[image]

I figure you are describing what i am seeing here. This is showing bridging over the channel for an o-ring groove

RetireeJay46, 2024-01-12:
Yes, themzlab, that's exactly what I mean.

sslupsky, 2024-05-23:
See also #7041

Jan-Soustruznik, 2024-05-28:
Issue closed as duplicate of https://github.com/prusa3d/PrusaSlicer/issues/7041