prusa3d/PrusaSlicer #11374: Extruder goes to the wrong spot on a Tool Change when using a Purge Tower

BrienKing, 2023-09-25:
### Description of the bug

When a Tool Change occurs the extruder is sitting at the last spot prior to a change as seen here:
[image]

When the new tool is undocked, it moves back to that spot, and then to the purge tower:
[image]

2 issues with that.  The first is that it can (and did) leave a bit of the wrong filament in that spot (even though it wasn't going to start printing from there).  The second is it wastes time as it has to move to the old location, then back to the purge tower and then back to the part again.   

It should go directly to the purge tower.

### Project file & How to reproduce

Create a multi-extruder model and look at the tool paths in Prusa Slicer at the end of a tool's use and the start of the next tool's use.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

Windows 10

### Printer model

Prusa XL Dual Extruder (4.7.2)

closed 2023-09-26 by kubispe1

BrienKing, 2023-09-26:
The problem isn't only in the GCode Viewer as it happens with the printer.  If it's a Firmware issue I can report it there, but the viewer shows the issue and it could be fixed via the slicer if necessary.   It could move to the purge tower then do a tool change.

I can provide a video later today if necessary.

BrienKing, 2023-10-03:
This appears to be a duplicate of this issue:
https://github.com/prusa3d/PrusaSlicer/issues/2672

themelle, 2023-10-26:
I assume issue [#9981](https://github.com/prusa3d/PrusaSlicer/issues/9981) may also be a direct duplicate, behavior seems identical.