prusa3d/PrusaSlicer #11378: Prusa XL Multi-Extruder Slicing changes color of first STL in one spot when it shouldn't

BrienKing, 2023-09-26:
### Description of the bug

I have 2 STL's that I imported into Prusa Slicer 2.6.1 and it asked if I wanted to join them, so I did.

Looking at it on the screen it looks correct as seen here:
[image]

When I slice it, it colors part of the red part in Yellow.
[image]

### Project file & How to reproduce

I was able to cut the part down to the minimum to show off the issue.
[Test Prusa XL Joust.zip](https://github.com/prusa3d/PrusaSlicer/files/12723619/Test.Prusa.XL.Joust.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

Windows 10

### Printer model

Prusa XL Dual Extruder (4.7.2)

kubispe1, 2023-09-26:
Quite a tough issue. 
- I would guess that this issue is related to wrong region detection. Where bridges somehow overflow to the middle part, where the top solid should be. Maybe some numeric issue.
- it is sensitive to print settings (only 0.40mm DRAFT does not work)
- PrusaSlicer-2.6.0-beta1 -  Doesnot work
- PrusaSlicer-2.6.0-alpha6 - still work
- Also I would guess that it is a less frequest issue

BrienKing, 2023-09-26:
I'll try a print at a lower layer height and see what happens for me.

BrienKing, 2023-09-26:
Seems to be an issue with the .4 Layer Height, works at lower layer heights.

BrienKing, 2023-10-03:
This appears to be a duplicate of this issue:
https://github.com/prusa3d/PrusaSlicer/issues/2672

BrienKing, 2023-10-03:
Posted to wrong issue, sorry.

closed 2026-09-29 by prusaslicer-bot[bot]