prusa3d/PrusaSlicer #11379: Crashes when slicing

mfraser, 2023-09-26:
### Description of the bug

PrusaSlicer crashes when slicing stl.
If I run it from the CLI, I get:
[2023-09-26 10:39:27.091238] [0x00007fb9062c8640] [error]   Reversing even wall line causes it to be printed CCW instead of CW!
Segmentation fault (core dumped)

### Project file & How to reproduce

1) Import STL file.
2) Click slice, PrusaSlicer gets to 45% and then crashes.
[Skeleton Feather Bowl.zip](https://github.com/prusa3d/PrusaSlicer/files/12725788/Skeleton.Feather.Bowl.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.6.1+linux-x64-GTK3

### Operating system

Kubuntu 22.04

### Printer model

Ender 3 V2 Neo

kubispe1, 2023-09-26:
For WIN 10:
PrusaSlicer-2.6.1+win64 (without cli) Crashes also to me.
PrusaSlicer-2.6.2-alpha1+63-win64 - looks ok. Let's check another platform. 
**
Such a nice model. I will launch it on Prusa XL. Let's see the result :D

mfraser, 2023-09-26:
> For WIN 10: PrusaSlicer-2.6.1+win64 (without cli) Crashes also to me. PrusaSlicer-2.6.2-alpha1+63-win64 - looks ok. Let's check another platform. ** Such a nice model. I will launch it on Prusa XL. Let's see the result :D

Is 2.6.2 alpha1 available to download?

kubispe1, 2023-09-26:
No, It is in the development stage, sorry. But it will come soon ... 
College detected it is a duplicate and should be already fixed... Here is the first one. #11267
Thanks, closing

closed 2023-09-26 by kubispe1