prusa3d/PrusaSlicer #11385: Paint On Fuzzy Skin

SloppiGoose, 2023-09-26:
I'm unsure if this has been brought up before, but paint on fuzzy skin would make a huge impact on the prints that I do. When I choose on outside walls only it still gets inside of my print and to set up modifiers so get everything that I need would be too time consuming. It would be set up the same way as paint on supports. You would just have 3 options for fuzzy skin, Paint, Every where, Outside. Giving you the option to use it or not. But still have it selected.

mix579, 2023-09-26:
Yes, it's has been requested many times before, consolidated under https://github.com/prusa3d/PrusaSlicer/issues/6988

Jan-Soustruznik, 2023-09-27:
Hello, @SloppiGoose , 

thank you for your feature request. We already know about this... We have it on our TODO list, but we have lot of more critical task that we have to do. I closed issue as duplicate. Thank you, have a nice day

closed 2023-09-27 by Jan-Soustruznik