prusa3d/PrusaSlicer #11390: PrusaSlicer crashes when loading Formfutura EasyFil PLA filament via Configuration Wizard

oxygene, 2023-09-28:
### Description of the bug

PrusaSlicer crashes 

### Project file & How to reproduce

Configuration --> Configuration Wizard --> Filaments
Select Type PLA, then select Vendor FormFutura, then FormFutura EasyFil PLA

Close the dialog with Finish button. Wait a couple of seconds, PrusaSlicer crashed.

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.1+win64

### Operating system

Windows 10

### Printer model

MK4 kit

rtyr, 2023-09-28:
I was able to reproduce it. This filament is part of the Anycubic bundle, but I dont see anything suspicious there.

lukasmatena, 2023-10-21:
Fixed in [PrusaSlicer 2.7.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-alpha1). Closing.

closed 2023-10-21 by lukasmatena