prusa3d/PrusaSlicer #11391: Post processing script not running win 10 / 2.6.1

Petri2, 2023-09-28:
### Description of the bug

After updating from 2.6.0 to 2.6.1 pyhthon post processing script stopped running.
I tried same script in 2.6.0 alpha and script works.
It also works when run from command line.
In 2.6.1 nothing happens. No error nothing.

The line on screen of slices Print Setting:
C:\Users\kesas\AppData\Local\Programs\Python\Python37\python.exe  "C:\Users\kesas\AppData\Roaming\PrusaSlicer\gcodefix\thumbnails.py";

in print.ini the following line exist:
post_process = "C:\\Users\\kesas\\AppData\\Local\\Programs\\Python\\Python37\\python.exe  \"C:\\Users\\kesas\\AppData\\Roaming\\PrusaSlicer\\gcodefix\\thumbnails.py\""

### Project file & How to reproduce

[kärrynpistoke-kansi.zip](https://github.com/prusa3d/PrusaSlicer/files/12752578/karrynpistoke-kansi.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1 + win64

### Operating system

windows 10

### Printer model

creality k1

Petri2, 2023-09-28:
This happens with any project and any printer even I attached project file.

Petri2, 2023-09-30:
Why this was closed? Not resolved!

Petri2, 2023-09-30:
Ok, I see I did it by myself! oops!

lukasmatena, 2023-10-04:
@Petri2 Your zip does not contain the script, and the 3MF that you enclosed does not have any post_process script configured. I have just run a Python post-processing script in 2.6.1 without any issues, so if you claim there is a problem, we need more info.

Petri2, 2023-10-05:
I responded already yesterday by email and also left here a comment. I wonder where my comment here was lost.

Investigating yesterday why you had not enough information although I tried to include all necessary I noticed another issue.
As I normally do not use project in prusa, normally I just import stl-files and do the slicing without saving any project,  this was not found before.
In my "print/output options" setup I have the script command but when I save a project and reopen it the "print/output options"  setup is blank.
I tried this several times and it seems always to be the case.
I suppose this is related to the issue.

I have the screenshots attached to my mail message.
[outputoptionscreenshots.zip](https://github.com/prusa3d/PrusaSlicer/files/12820741/outputoptionscreenshots.zip)

The "print/output options"  setup line  is currently as follows:
C:\Users\kesas\AppData\Local\Programs\Python\Python37\python.exe  "C:\Users\kesas\AppData\Roaming\PrusaSlicer\gcodefix\thumbnails.py";

So this is not saved (or at least lost) in saving as a project.
Currently I run this script manually on command line to manage thumbnails which do not othervise work because Crelity is not using same thumbnail header as Prusa. (well, that is another issue, but I believe it is also known).

closed 2026-09-29 by prusaslicer-bot[bot]