prusa3d/PrusaSlicer #11392: Bug in the arrange algorithm - hangs at 60%

kamikasee, 2023-09-28:
### Description of the bug

I was trying out the auto-arrange these two models, and the slicer just hangs forever at 60%.  The two triangle shapes seem like they should be straightforward to arrange, so I expected it to complete.

Here are the auto-arrange settings I was using.
[image]

  I can arrange these two items by hand pretty easily, so this bug is not a blocker.  I'm submitting the bug in case it makes a useful test case or identifies an unusual corner case for the arrange algorithm.

### Project file & How to reproduce

[arrange-bug-report.zip](https://github.com/prusa3d/PrusaSlicer/files/12754633/arrange-bug-report.zip)
[arrange-bug-config.zip](https://github.com/prusa3d/PrusaSlicer/files/12754645/arrange-bug-config.zip)

Have the two triangular pieces with Printable turned on, the rest with Printable turned off.
Click the auto-arrange button
See the algorithm get to 60%
I waited about 10 minutes before giving up and cancelling.

If you need more information, feel free to reach out.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1+win64 PrusaSlicer-2.6.1+win64-202309060711

### Operating system

PrusaSlicer-2.6.1+win64-202309060711

### Printer model

Prusa MK3S+

lukasmatena, 2023-10-04:
Thanks, reproduced. We will look into it.
SPE-1934

closed 2023-11-16 by lukasmatena

lukasmatena, 2023-11-16:
Fixed in [2.7.0-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-rc1).