prusa3d/PrusaSlicer #11393: "Thick Bridges" does not respect "Layer Height Limits" (even when thick bridges is disabled)

LWooler, 2023-09-29:
### Description of the bug

"Thick Bridges" does not respect "Layer Height Limits"
I set my layer height max to 0.32, my nozzle size it 0.4, and enable thick bridges.
Generate gcode
Looking at the render, the bridging layer height is 0.39

I set my layer height max to 0.32, my nozzle size it 0.4, and enable thick bridges.
Generate gcode
There are still bridging infill layers printing at 0.39.

### Project file & How to reproduce

[thick_bridges_not_respecting_layer_height.zip](https://github.com/prusa3d/PrusaSlicer/files/12755838/thick_bridges_not_respecting_layer_height.zip)

[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1 +win64

### Operating system

Win 10

### Printer model

Ender 3 Pro -> Klipper/Mainsail

rtyr, 2023-09-29:
> "Thick Bridges" does not respect "Layer Height Limits"

This is standard/intended behavior. When the "thick bridges" option is enabled, the bridge flow is based on nozzle diameter, not on layer height. 

See:
https://help.prusa3d.com/article/layers-and-perimeters_1748#thick-bridges

https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.0-alpha1

https://manual.slic3r.org/advanced/flow-math

closed 2023-09-29 by rtyr

eduncan911, 2024-07-05:
@rtyr please re-open. Your understanding is incorrect in this manner.  Let me try to explain the problem another way...

The issue is **_Layer Height_** is increased to 0.6mm (on my 0.6mm nozzle).  Period.

Thick Bridges is DISABLED !!

That's the bug.  Regardless of what I or the OP turns off for thick bridges, IT USES OUR NOZZLE WIDTH AS HEIGHT.

I'm using the latest PS 2.8.0 btw.

[image]

rtyr, 2024-07-06:
Bridges generated over sparse infill always use "thick bridges" flow.

eduncan911, 2024-07-11:
> Bridges generated over sparse infill always use "thick bridges" flow.

Again, you are dismissing this without understanding. There is no spares infill!  You are wrong to state that.

Please reopen.

eduncan911, 2024-07-11:
> Bridges generated over sparse infill always use "thick bridges" flow.

Than that's the bug.  Please reopen so we can address it.

Please do not continue to dismiss the FLAW where other slices do NOT.