prusa3d/PrusaSlicer #11394: Slicing Crashes on Large Objects

blakeglucas, 2023-09-29:
### Description of the bug

Slicing the Creality Rabbit for the Creality CR-M4 printer (https://www.thingiverse.com/thing:5757930/files, rabbit_009_.stl), PrusaSlicer crashes with no warning dialog. Because the CR-M4 printer is a large printer, I've scaled the rabbit STL to 500%. Project and config files are in the attached ZIP file. No error or warning is given, the application just crashes.

### Project file & How to reproduce

[rabbit_009__bug_report.zip](https://github.com/prusa3d/PrusaSlicer/files/12755902/rabbit_009__bug_report.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1+win64

### Operating system

Windows 11 Home

### Printer model

Creality CR-M4

blakeglucas, 2023-09-29:
Update: it works for scaling to 399%, but 400% causes the same crash. No dialog, logs, nothing.

MyChaOS87, 2023-10-01:
Just tried it out of curiosity... for me it works, even with 800% multiple tries now. 

But on 800% I get serious load on CPU (100%) and ~6GB RAM usage. (Core i7-12700H and 64gb RAM)

Perhaps you should check ressorce usage on your machine and check your machine under load as well

blakeglucas, 2023-10-06:
I'm assuming there was some sort of load on my machine or GPU, because now I can't reproduce it either. Closing.

closed 2023-10-06 by blakeglucas