prusa3d/PrusaSlicer #11396: OpenGL not hardware-accelerated over RDP

apirek, 2023-09-29:
### Description of the bug

When starting PrusaSlicer in a RDP session, the llvmpipe OpenGL driver is loaded instead of the appropriate hardware driver (Nvidia in my case).
Other OpenGL softwares (e.g. PreForm) are using hardware acceleration.
This happens with FreeRDP version 2.11.2 on Arch Linux (the `/rfx` flag makes no difference) and with the built-in remote desktop client on Windows 10.

If I force PrusaSlicer to use hardware acceleration with the undocumented `--no-sw-renderer` flag it is working without problems.

### Project file & How to reproduce

1. Connect via RDP
2. Start PrusaSlicer
3. Check Help -> System Info, see Renderer: Gallium 0.4 on llvmpipe instead of my graphics card

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1+win64

### Operating system

Windows 10

### Printer model

any

apirek, 2023-12-02:
Still happens with PrusaSlicer 2.7.0.

I tried forcing hardware acceleration with some more Windows computers, it works on all of them:
* Windows 10 Version 21H2 (Build 19044.3086) with NVIDIA Quadro K2000, driver version 10.18.13.5330 (NVIDIA 353.30)
* Windows 10 Version 22H2 (Build 19045.3693) with AMD Radeon RX550, driver version 30.0.13023.4001 (Adrenalin 21.30.23.04) DCH
* Windows 11 Version 22H2 (Build 22621.2715) with Intel UHD Graphics 620, driver version 31.0.101.1999 DCH

@bubnikv introduced a check for RDP in addition to the OpenGL version in https://github.com/prusa3d/PrusaSlicer/commit/88059baddba34fafd9952775491732192999fb3d in 2019 but there is no explanation as to why.
According to the comment above the check it would be redundant because the version check handles that case already.

But that comment holds no longer true. I get the hardware OpenGL 4.x everywhere out of the box, I did not have to configure any group policies[^1] or run some strange tool[^2].

[^1]: https://community.esri.com/t5/implementing-arcgis-questions/enabling-gpu-rendering-on-windows-server-2016/td-p/658522
[^2]: https://www.khronos.org/news/permalink/nvidia-provides-opengl-accelerated-remote-desktop-for-geforce-5e88fc2035e342.98417181

ApexArray, 2024-03-11:
Oh man, I'm so glad I found this post. I use PS through RDP 99% of the time and the G-Code preview is pretty much unusable when viewing more than a few layers at a time. I thought I had ruled out RDP as the issue since it was still slow after logging into the physical desktop (using the same PS instance started over RDP).

I confirmed the issue is fixed when starting PS from the physical desktop, or by adding the mentioned `--no-sw-renderer`. It feels like a completely different app now.

+1 for removing this check from the code, or at least giving an option to toggle hardware acceleration in the options menu.

<details>
<summary>Hardware specs</summary>

Windows 11
Ryzen 3600
GTX 1080 TI
PS 2.7.2 (going back at least as far as 2.5)

</details>

apirek, 2024-03-12:
I installed PrusaSlicer 2.7.2 on an old laptop computer with Windows 7 Version 6.1 (Build 7601: Service Pack 1) and Intel HD Graphics 4000, driver version 10.18.10.4425, no updates except KB2999226 which was required to get PrusaSlicer to start.
Windows 7 is the oldest Windows still supported by PrusaSlicer.
The GPU supports OpenGL 4.0 and PrusaSlicer uses hardware acceleration when started locally.

On this version of Windows and RDP, overriding the RDP check with the `--no-sw-renderer` flag has no effect, it **does not break anything** and PrusaSlicer just uses Mesa.
Other OpenGL softwares (e.g. PreForm) are not using hardware acceleration either.

1n5aN1aC, 2024-12-20:
Been having this same problem for years, finally figured out the problem thanks to this bug.

Can report exactly the same.  PrusaSlicer GCODE previews can sometimes lag the client so bad it completely locks up when starting the program over RDP.

Windows 10, Ryzen 3600, Nvidia 4070.

I can also confirm launching the program with `--no-sw-renderer` does resolve the issue for me as well.

Batailic, 2025-09-29:
This still isn’t fixed in the latest 2.9.3 version making working remotely quite frustrating…
Maybe just add configuration parameter to control software rendering over RDP instead of forcing the use of obscure command line parameter…

tasols, 2026-02-10:
I thought OpenGL hardware acceleration was impossible in RDP sessions. After seeing that Bambu Studio works properly even in RDP sessions, I searched for a solution and ended up here.
Thank you! With the --no-sw-renderer flag, I can now use it smoothly.

However, I'm quite disappointed to find that PrusaSlicer—the original project—has this RDP detection issue, while Bambu Studio, its fork, does not. It's ironic that the derivative handled this better than the source.

apirek, 2026-09-01:
No longer an issue in PrusaSlicer 3.0.0-alpha11. I can't find the System Info dialog anymore, but running with `--opengl-debug` confirms it is using my GPU.
I'll await the stable release before closing this issue.

closed 2026-09-29 by prusaslicer-bot[bot]