prusa3d/PrusaSlicer #11399: PrusaConnect Issues

JakobGlieder, 2023-09-30:
### Description of the bug

When I try to upload something to prusaconnect with Prusa Slicer, It uploads just fine at first but them gets stuck at 100% for some time, until I get an error. Connection works according to the test. After trying it again a few times it usually works.

### Project file & How to reproduce

1. Have a printer Connected to PrusaConnect, and a Physical Prusaconnect Printer set up in PrusaSlicer.
2. I used the InputSlicer (Stable, not Alpha) Profile, don't think that makes a difference but not sure.
3. Try uploading any sliced file from Prusaslicer to connect, no matter if just upload, que, or print.
[image]
[image]
[image]
PrusaSlicer_config_bundle
[report.zip](https://github.com/prusa3d/PrusaSlicer/files/12775108/report.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.1+MacOS-arm64

### Operating system

13.6 (22G120)

### Printer model

Original Prusa MK4 Kit

Jan-Soustruznik, 2023-10-02:
Hello, @JakobGlieder , 

Thank you for your report. We already know about this issue. Problem is on PrusaConnect side. We already informed connect team, they will try to solve. 

I closed this issue a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/11076 

Have a nice day

closed 2023-10-02 by Jan-Soustruznik

lukasmatena, 2023-11-11:
There is an update in https://github.com/prusa3d/PrusaSlicer/issues/11076, we would be grateful if you also had a look. Thanks.