prusa3d/PrusaSlicer #11400: Odd object on the bed

StenPetrov, 2023-09-30:
### Description of the bug

There's some odd object that keeps appearing in Prusa Slicer 2.6.1, after I upgraded. 
It doesn't print and can not be removed. There are no properties shown (position, size, rotation, scale) when the object is selected.
It is not an object I have made or imported, I don't recognize it. It shows up when I open Prusa Slicer. It keeps showing after I uninstalled, deleted all folders named "*prusa*" and installed again. On another computer where I have Prusa Slicer it doesn't show up.

[image]

### Project file & How to reproduce

Just open Prusa Slicer, the odd object shows up on the bed

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

Windows 10

### Printer model

Prusa XL and Prusa Mk3S

rtyr, 2023-10-01:
It's normal. It is visual representation of the wipe tower. It's printed only if there are some tool changes.

closed 2023-10-01 by rtyr

StenPetrov, 2023-10-01:
@rtyr This makes no sense, since the printer is PrusaXL and it doesn't create wipe towers

rtyr, 2023-10-01:
> is PrusaXL and it doesn't create wipe towers

Prusa XL does create wipe towers, even though their purpose is mainly to pressurize the nozzle. 

> Also the shape is way too specific to be a wipe tower, which should be rectangular

It is a wipe tower. The cone shape is to make it more stable with higher prints.

rtyr, 2023-10-01:
If you are using single tool XL, select single tool XL in the wizard (you are using 2 tool profiles based on the screenshot). You can disable wipe tower in print settings if you want, but it is enabled by default in standard profiles for obvious reasons.

rtyr, 2023-10-01:
See the chapter 4.8 in Prusa XL handbook or you can contact our tech-support via live chat at www.prusa3d.com.