prusa3d/PrusaSlicer #11404: URI Too Long

chrishilbert, 2023-10-02:
### Description of the bug

Prusa Slicer will attempt to send a too long filename to Prusa Link machines.  It will catch the below error message. 

[image]

Prusa Slicer really should do something like:

1.  Limit the text field for the filename to the max length.
2.  Have an option on the dialog box to strip off all the extra parameterized stuff - nozzle size, material, etc.
3.  Suggest a trimmed filename.

### Project file & How to reproduce

Make a really log file name and click "upload and print".

For example:  A Really Long File Name That Throws An Error_0.4n_0.2mm_PLA_MINIIS_41m.gcode

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

Mac

### Printer model

Prusa Mini with IS Alpha

jmd73, 2023-10-02:
@chrishilbert  Duplicate of #10901

rtyr, 2023-10-02:
Yes. The message is returned by the firmware.

closed 2023-10-02 by rtyr