prusa3d/PrusaSlicer #11405: Error on startup: SuperSlicer - Switching language failed

dpd123, 2023-10-02:
### Description of the bug

I'm attempting to run SuperSlicer for the first time.

"SuperSlicer - Switching language failed.
You may need to reconfigure the missing locales, likely by running the "locale-gen" and "dpkg-reconfigure locales" commands."

I see some references to similar issues from a couple years ago, seemingly fixed. Any ideas to get SuperSlicer to start?

### Project file & How to reproduce

Nothing to show, program won't start.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

SuperSlicer_2.5.59.2+UNKNOWN based on Slic3r (with GUI support)

### Operating system

Gentoo Linux

### Printer model

FLSun V400, Ender 3V2

rtyr, 2023-10-02:
This is PrusaSlicer repository.

closed 2023-10-02 by rtyr

jmd73, 2023-10-02:
@dpd123 
Wrong repo. You should report this in [SuperSlicer repo](https://github.com/supermerill/SuperSlicer)

dpd123, 2023-10-10:
Hello, sorry for my confusion.
The problem also happens with PrusaSlicer:

[image]

PrusaSlicer-2.5.2+Gentoo based on Slic3r (with GUI support)

Any ideas?

dpd123, 2023-10-10:
Sorry for the outdated version in the report.
However prusaslicer-2.6.0-r1::gentoo ebuild also has the same problem.

dpd123, 2023-10-10:
Sorry for quoting the outdated version, the Gentoo repo is a bit behind.
I also tried both GTK2 and GTK3 AppImage releases, same problem.

dpd123, 2023-10-10:
I was able to get prusa slicer to launch by doing the following:

1. Add en_US UTF-8 locale to the system:
> localedef -i en_US -f UTF-8 en_US.UTF-8

2. Launch with the command:
> LC_ALL='en_US.UTF-8' ./PrusaSlicer-2.6.1+linux-x64-GTK3-202309060711.AppImage

It did NOT work when attempting to launch with the Canada locale:
> LC_ALL='c.UTF-8' ./PrusaSlicer-2.6.1+linux-x64-GTK3-202309060711.AppImage