prusa3d/PrusaSlicer #11406: Emboss recycling symbols

ciaranmcnulty, 2023-10-02:
I'm currently adding recycling symbols to prints using Parts and the gallery. 

However, moving a part around to get the right distance of emboss / deboss is pretty manual. It'd be great to be able to use the new Emboss feature's ability to follow normals, especially on curved surfaces.

It'd be great to be able to select a location on the print, a depth/height, and orientation the way we can for text.

ciaranmcnulty, 2023-10-02:
Related is https://github.com/prusa3d/PrusaSlicer/issues/9439 which asks for a more generic logo emboss.

I think this feature is simpler and could be a limited set of embossable logos built in

foreachthing, 2023-10-02:
I've "created" a whole bunch of symbols here https://github.com/foreachthing/RecyclingSymbols.
These can be added as negative volume...
Maybe add these (or some) symbols as font elements?!

closed 2026-09-29 by prusaslicer-bot[bot]