prusa3d/PrusaSlicer #11409: Cannot export a part as an stl correctly that was modified with a negative volume.

VH-fixer, 2023-10-02:
### Description of the bug

I have assembled a part and added a hollow cylinder as a negative volume.   When I try to save as an updated .stl, I get an error and it tells me only the positive part will be saved.   
[image]

### Project file & How to reproduce

[Universal_fin-clipped.zip](https://github.com/prusa3d/PrusaSlicer/files/12786871/Universal_fin-clipped.zip)

Open the project.  Select the top level assembly.  RMB and select the Export stl/obj option.  The error appears. 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

Windows 11 22H2

### Printer model

MK4

mcmusic2, 2024-08-01:
I just wanted to file the same bug. See attached pic. I also used a hollow cylinder as a tool.
[image]

AndrewFarley, 2024-11-17:
Can we please get this feature?  Or, can someone who knows some history of why this isn't in the Prusa Slicer comment on why we don't?  I'd like to understand, and consider undertaking adding this feature myself if I can but I want to jump into this informed.