prusa3d/PrusaSlicer #11410: A toolpath outside the print area was detected

BrienKing, 2023-10-03:
### Description of the bug

When the slicer automatically places the purge tower, it is too close to the edge and gets that error.

It's either an error in the detection because it's right on the line, or an error on the automatic placement that allows it to be put outside the printable area.   There is plenty of room on the bed.

[image]

### Project file & How to reproduce

See the picture above.    Should move the parts/purge tower further from the line.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

Windows 10

### Printer model

Prusa XL Dual Extruder (4.7.2)

lukasmatena, 2023-10-03:
Seems like a duplicate of #11367. Unfortunately, neither of you included a 3MF. Can you please provide it? Thanks.

BrienKing, 2023-10-03:
Here you go.

[super_star.zip](https://github.com/prusa3d/PrusaSlicer/files/12789967/super_star.zip)

lukasmatena, 2023-10-03:
@BrienKing Thanks, but this is only the STL. We are unable to reproduce the issue and we need a full 3MF project with your configuration (File->Save project), because something in the configuration can trigger it.

You may also include a screenshot of the settings in the arrange dialog (which opens by right clicking the arrange icon in the top toolbar), because those are not saved in 3MFs.

BrienKing, 2023-10-03:
Sorry about that.  This is the 3MF file.

[super_star.zip](https://github.com/prusa3d/PrusaSlicer/files/12794022/super_star.zip)

FloatingCam, 2024-02-04:
I am having the same problem on my Prusa XL5.  I load an STL, click Arrange, and when I slice it this error is presented. 

I found, when it happens, changing the setting under Print Settings->MultipleExtruders->Wipe tower brim width: from 3 to 2 fixes it.  The wipe tower is being 'Arranged' off the plate due to its brim.

While not a serious bug **this is annoying and should be an easy fix.**  I've seen this several times and only had the printer a couple of days. This is not difficult to recreate.  With a multi-color print (4 or 5 colors) it seems to happen a lot.  

An NO, you don't need my STL to fix this.  If this isn't enough information I cannot believe sending you another STL is going to make a difference.

mr-ice, 2024-02-05:
I think this is the same as in #11367

It has been several months, are there any updates @lukasmatena?

On #11367 I provided two 3mf files on that other ticket, one with the automatically assigned position, demonstrating the toolpath outside the print area.  The other with a manually adjusted wipe tower 'y' setting that shows that moving that in 0.2~0.5mm does not show the same error.  I think that effectively demonstrates that the wipe tower position is the problem.

here are my arrange settings:
[image]

lukasmatena, 2024-02-09:
Fixed in [2.7.2-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.2-alpha2). Closing.

closed 2024-02-09 by lukasmatena