prusa3d/PrusaSlicer #11412: Allow slicing of oversize parts when an added negative volume will keep the result within printing limits.

VH-fixer, 2023-10-03:
There are times I want to print parts that initially are too big for the printer.  By adding negative volumes the resulting part fit the limits.   Prusa Slicer will not allow this to be sliced.  The positive part outside limits prevents slicing.   I turned it in as a bug, but was told it works as intended.  Boleans are done after slicing.    
This part was designed to be scale to fit, but scaling it up to fit pushed the fins outside of the MK4 limits.  That is why I added the negative volume.   This is a screenshot of the error.  

[image]

As a workaround I tried to export the resulting model as an .stl, but I got an error and it said the negative volume would not be applied.   I turned this in as a bug.  

Cutting is not a good option, it would be many more steps and I don't want just flat planes as a result.   In this example the resulting edge would be curved.  Getting all three edges with the same length is nearly impossible for most users.  

Prusa slicer with the cut and negative volume capabilites already has the functions to do this.  The code just needs to be adjusted.   The solutions I see are:
1. Use cut bolean functions to remove the negative volume and only have the final part analyzed for slicing.   Then check slicer limits to allow slicing.    Calculating a new  part or .stl from negative volumes is new and still needs to be made much more reliable. 
2.  Prusa slicer needs to calculate slicing boundaries with the negative volume applied.  Slicing should be allowed even when a non-negative piece of the part is out of bounds.    
3.  A user override for slicing warnings or errors.    Allow slicing even with an error.  Then after slicing is finished report if there is still an out of bounds move.

jonatino, 2024-05-05:
I wish there was an option to not exclude out of bounds objects automatically. I understand why that is the default behavior, but I should be able to override the system and print anyways and it can just ignore out of bounds coordinates.

VH-fixer, 2026-09-19:
I don't see this being addressed in the 3.x list.