prusa3d/PrusaSlicer #11415: connectors dissapear when slicing => only when using multiple cut planes AND excluding parts

paloss, 2023-10-04:
### Description of the bug

Whats the problem: connectors added by slicer when plane cutting are visable in the model but dissapear after slicing 

what did you expect: connecters still there after slicing

This ONLY happens when there is a first cut WITH connector, then another cut WITH connector AND 1 part excluded in the cutting plane. If any of these are different it works fine. Its very specific 

It happend in my 3d model that i needed to split in 3 parts but i am able to recreate it with just adding "shapes" in slicer.

### Project file & How to reproduce

Step 1: add 4 bloks, merge them and stack them as in the attached picture
step 2: make a low plane cut with 1 connector.
step 3: make a high plane cut with 1 top side apart, add a connector to the right big part
result=connector visable in the model => slice => connector is there

ctrl z, edit the cut, only right click the "floating" top part of the left square. 
result=connector visable in the model => slice => connector gone.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[blocks connector bug.zip](https://github.com/prusa3d/PrusaSlicer/files/12807171/blocks.connector.bug.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Versie 2.6.1+win64

### Operating system

win 11 latest

### Printer model

mk4

YuSanka, 2023-10-05:
Thanks for your report. (SPE-1938)

bgyarfas, 2023-10-13:
I also noticed the crash when trying to use the dowel and right-clicking a part.

lukasmatena, 2023-10-23:
The crash reported by @bgyarfas is fixed in 2.7.0-alpha1. The original issue will be tracked in #11487.