prusa3d/PrusaSlicer #11421: Incorect automatic arrangment

rafibar, 2023-10-05:
### Description of the bug

[image]
this is the automatic arrangmet for those parts.
after slicing i have an error message and the supports overlap on each other.
[image]

### Project file & How to reproduce

[side.zip](https://github.com/prusa3d/PrusaSlicer/files/12819451/side.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1 64bit

### Operating system

win 11

### Printer model

mk3s+

lukasmatena, 2023-10-06:
This is a duplicate of #10486. Unfortunately, this has no simple solution. Before the supports are actually generated, it is not possible to tell how far from the object they will extend. So far the only thing that you can do is to right-click the arrange button in the top toolbar and manually increase the spacing between objects.

Closing as a duplicate.

closed 2023-10-06 by lukasmatena

rafibar, 2023-10-06:
thanks for your reply