prusa3d/PrusaSlicer #11423: Multiple Wipe Tower for using PLA/PETG support combo.

jfmmm, 2023-10-05:
**Is your feature request related to a problem? Please describe.**
When using PLA as support for a PETG print or the opposite with a dual tool XL, the PLA won't stick to the PETG on the wipe tower and just create a mess that then get deposited on the print. This happen even when you set a specific extruder for the wipe tower.

**Describe the solution you'd like**
There should be an option to allow multiple wipe tower to be used for specific extruder. Or at least an individual wipe tower per extruder.

jfmmm, 2023-10-06:
Just realized that you also can't use the prime tower with a different size nozzle if you are trying to use a bigger nozzle for infill.

Another place where having the option to have more than one wipe tower would help a lot.

Braintoe, 2023-10-06:
Is second this issue. PETG makes a great removable support for PLA and vice versa (the main reason for me buying an IDEX printer), but in opposite to e.g. the Cura wipe tower, the PrusaSlice Wipe tower becomes unstable with this combination since it prints each layer with another hotend. You can counteract this somewhat by creating very large cone and increase wipe tower extra spacing to get a larger footprint, but this is limited.

This issue applies not only to PLA and PETG but to all support materials that are meant to come off easily after the part has been printed.. 

The Cura way of creating a wipe tower (an example can be seen here: https://core-electronics.com.au/guides/3d-printing/purge-towers-ooze-shields/ ) is much better in this respect - they use one perimeter for each extruder and have a hole in the centre of the wipe tower. 
In the case of two extruders, the extruder printing the outer perimeter (let's assume it is the first extruder) always starts printing coming from the outside, whereas the second extruder that prints the inside of the wipe tower will always start printing its perimeter coming from the inside. This means any excess material dangling from the first extruder - e.g, from an unretract - will form a blob on the outside of the wipe tower, and any excess material from the second extruder will form a blob on the inside of the wipe tower. Additionally, to keep such blobs from interfering with the next layer, Cura moves the start point of each extruder on the wipe tower with each layer.

Braintoe, 2023-10-07:
A possible solution how this could be implemented would be to use the wipe tower as it is but add two options:
1. a Filament option "adhesion" ranging from e.g. 1...16 where the user defines the adhesion quality of this material relative to all the other materials he uses - the higher the value the better.
2. in addition to wipe_tower_width", a second option "wipe_tower_depth" which would then allow to create a hollow wipe tower instead of the massive one.

Behaviour:
- if two materials have the same adhesion value, Prusaslicer assumes these are compatible and keeps the current behaviour of the wipe tower.
- if two materials have a different adhesion value, Prusaslicer assumes these will probably not stick together and uses separate perimeters of each wipe tower layer for these, thereby ensuring that these materials will never have to stick on each other. For example, if extruder 1 uses PLA with a user-defined adhesion of 8 and extruder 2 uses PETG with a user-defined adhesion of 5, extruder 1 starts creating the wipe tower layer beginning from the outside as it uses the material with the highest adhesion coefficient. Then, the next inner part is created by the extruder which uses the material with the with the next lower adhesion value, in the example extruder 2. (If there would be another extruder with another material with adhesion below 5, this one would create another inner part of the wipe tower). Since this requires all extruders except the first to dive into the centre of the purge tower (otherwise the outer part which is printed by extruder 1 could be in the way) this ensures that excess material from extruder 1 will be on the outside of the purge tower and excess material of the following extruders will be on the inside.

Wipe_tower_depth would be set if the user wishes to create a larger (and therefore hollow) wipe tower.
- if the value is not set or too low, the wipe tower behaves as it currently does, the depth is calculated from wipe_tower_width and the purge volume.
- if the value is above the calculated depth, the wipe tower becomes hollow. If extruders with a lower adhesion than extruder 1 value exist, these dive into the centre of the wipe tower and then move to where they start creating "their" part of the perimeter.

murk-sy, 2023-10-23:
There will definitely be cases where 2 different materials are used in addition to the support material, so having the option of multiple prime towers *may* be necessary, but I haven't tested that yet.

For now, marking PLA (or whichever is the support) as `Soluble material` will prevent Prusaslicer from depositing it inbetween the prime tower layers.

Normal:
[image]

Purple marked as soluble:
[image]

Braintoe, 2023-10-23:
@murk-sy that works only partly, see here:
[image]
[image]

PETG (marked as "soluble") is dark grey in hotend 1, PLA is light grey in hotend 2. 
Despite "wipe_tower_extruder" being set to "0" which - if I understand the hint text "The extruder to use when printing perimeter of the wipe tower. Set to 0 to use the one that is available (non-soluble would be preferred)." correctly should mean "nonsoluble preferred", Prusaslicer does print layer 1 with PETG only - which means the wipe tower in itself is stable, but its base is - sadly - not. If I interpret https://3dfans.de/prusaslicer-2-4-0-alpha1-released/ correctly, this seems to be intended behaviour which I guess makes sense for soluble material but not that much for PETG.
Apart from that, the same page says that the "soluble" switch also affects other things apart from the wipe tower... while the things mentioned there totally make sense for soluble material, they again do not really do for PETG support material...

dattrax, 2024-03-15:
I've just had a prime tower fail on petg + pla.  I was able to stand over the print and make sure it didn't blob out, but its not something I enjoy doing for 3hrs.  It was a 36hr print, so I was going to make sure it finished.

Can I +2 for this feature.

77erup, 2025-04-11:
I'm definitely in.  Having the ability to create 2 wipe towers for PLA and PETG prints would be very helpful.

dsanders2, 2025-04-22:
I was able to get around this problem by setting the Wipe tower extruder. I set it to the most used nozzle (to save filament). It creates the perimeters as PLA and then the interior is just lines of the other filaments. Worked great for me.