prusa3d/PrusaSlicer #11425: Ender 3 S1 Pro Bed Shape

DjordjeMandic, 2023-10-05:
### Description of the bug

Printable area is 215x220, actual bed is 235x235. Shouldnt printable area be 220x220 and purge line at coordinates like Y-2

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

Windows 10

### Printer model

Ender 3 S1 Pro Stock

rtyr, 2023-10-06:
@pmjdebruijn fyi

pmjdebruijn, 2023-10-06:
@DjordjeMandic did you verify all physical bed area is actually addressable? i.e. the nozzle can reach it?

DjordjeMandic, 2023-10-06:
@pmjdebruijn https://github.com/prusa3d/PrusaSlicer/assets/6750655/fdb55686-1526-4e30-a4c2-b2f69d051a13

pmjdebruijn, 2023-10-07:
I won't be accepting any custom per-printer gcode's though, especially not for a few millimeters of gain, as that will result in a printer bundle maintenance nightmare.

DjordjeMandic, 2023-10-07:
Is bed set to 215x220 because of purge line being at x 0 or something else?

pmjdebruijn, 2023-10-07:
I don't recall this case specifically, but that's likely yes.

pmjdebruijn, 2023-10-07:
@rtyr PrusaSlicer still has no mechanism to have per-printer custom variables does it? so I can make the purge location configurable without custom gcode's ?

rtyr, 2023-10-09:
no

closed 2026-09-02 by DjordjeMandic