prusa3d/PrusaSlicer #11426: PrusaSlicer 2.6.1 hangs while "Making Supports"

konkers, 2023-10-05:
### Description of the bug

When trying to slice the below file with the `0.20mm SPEEED` profile for an `Original Prusa MK4 Input Shaper 0.4 nozzle` printer profile, the process hangs at 45% while making supports.  If I increase the vertical shells to 3, the model correctly slices.

### Project file & How to reproduce

[Body1.zip](https://github.com/prusa3d/PrusaSlicer/files/12825031/Body1.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

Windows 10

### Printer model

Prusa MK4

lukasmatena, 2023-10-06:
Thanks, reproduced. We will look into it.
SPE-1950

lukasmatena, 2024-09-12:
Fixed in [2.8.1-rc1](https://github.com/prusa3d/PrusaSlicer/releases). Closing.

closed 2024-09-12 by lukasmatena