prusa3d/PrusaSlicer #11427: Holes disappear after slicing the model

xiaojie0815, 2023-10-06:
### Description of the bug

Holes disappear after slicing the model

### Project file & How to reproduce

[image]
[image]
[零件3.zip](https://github.com/prusa3d/PrusaSlicer/files/12826307/3.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

3.6.1

### Operating system

win10

### Printer model

voron2.4

jmd73, 2023-10-06:
@xiaojie0815 It slices fine for me. Can you provide a 3mf file instead for the STL?

I suspect it is some setting that are affecting this.

lukasmatena, 2023-10-06:
I would check `Print Settings -> Advanced -> Slicing mode`. If that's not the problem, we will really need the 3mf.

xiaojie0815, 2024-01-11:
> @xiaojie0815对我来说切片很好。你能提供一个 3mf 文件来代替 STL 吗？
> 
> 我怀疑是某些设置影响了这一点。
I'm really sorry. Because my company is busy with overtime work, I forgot to reply. I found that this bug has been fixed in PrusaSlicer2.7.1 version and cannot be reproduced. I'm sorry to trouble you. Thank you for your hard work and answers.

closed 2026-09-29 by prusaslicer-bot[bot]