prusa3d/PrusaSlicer #11428: MMU: fill up embossed letters so that I can print them in a different color

pathia, 2023-10-06:
**Is your feature request related to a problem? Please describe.**
Whenever downloading models that someone else made who doesn't have a MMU they always emboss letters to make them visible. However since I have a MMU3 I want to fill the embossing with another color for a much nicer look.

**Describe the solution you'd like**
Make it possible to fill embossed letters with another color.

**Describe how it would work**
I would like it to be possible that I click letters in my model and then something like 'fill' and then the filled part would be a different object/part so that I can click the part in the list on the right and select the extruder color.

**Describe alternatives you've considered**
I tried editting the stl file in Fusion but I would need a expensive license to do this. Also it would be much more convenient if this is already built in in Prusaslicer.

**Additional context**
Add any other context or paint drawing about the feature request here.

jmd73, 2023-10-06:
@pathia You can use the painting functionality in PrusaSlicer for that.

See the [PrusaSlicer manual](https://help.prusa3d.com/article/multi-material-painting_262620) for more details.
 
Exampl:
[image]

pathia, 2023-10-06:
Whoops I meant debossed letters. Then you would get this effect:
[image]
I'll correct the title of this FR.

FidelCapo, 2023-10-09:
@pathia I'm not sure if I understood the request correctly, but if you create the text as a modifier and assign it a different extruder color in the right panel, does it create what you want?

pathia, 2023-10-09:
> @pathia I'm not sure if I understood the request correctly, but if you create the text as a modifier and assign it a different extruder color in the right panel, does it create what you want?

Well, it could be but then I would have to select the exact same font and size and position it exactly in the gaps of the debossing.

FidelCapo, 2023-10-09:
Would a copy/paste of the text volume help? Is the debossed text created by you in the slicer or is it already part of the imported object? Alternatively, can you send me a 3mf file of the project so I can better understand what you are trying to achieve?

pathia, 2023-10-09:
> Would a copy/paste of the text volume help? Is the debossed text created by you in the slicer or is it already part of the imported object? Alternatively, can you send me a 3mf file of the project so I can better understand what you are trying to achieve?

For example take the top lid of this box: https://www.thingiverse.com/thing:5251883/files
The letters are debossed into the lid which is the best you can do with single material printers but I want to do the letters in a different color so I need the debossed part to be filled with material and as a different object inside Prusaslicer.

pathia, 2023-10-09:
> Oh, I see. Using the part and the correct order might help. Please look at the attached project.
> 
> [DT3top-MMU_Text.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/12843687/DT3top-MMU_Text.3mf.zip)

That seems to be what I want to do. How did you do that?

FidelCapo, 2023-10-09:
I added a part to the downloaded object in the slicer (right mouse click on the object - add part - Slab), changed the size and position of the Slab, changed the order of the parts in the right panel (to give the object a higher priority) and assigned a different extruder to the added part.

pathia, 2023-10-09:
> I added a part to the downloaded object in the slicer (right mouse click on the object - add part - Slab), changed the size and position of the Slab, changed the order of the parts in the right panel (to give the object a higher priority) and assigned a different extruder to the added part.

I tried something like this but had no idea about the order of the parts and how to rearrange them. This really helps.
Thank you for the help! 👍

Nevertheless, I think this would still be a awesome feature to directly integrate inside Prusaslicer. Just pick the letters and automatically they will be filled up with a different part and select the extruder color for it.

FidelCapo, 2023-10-09:
I'm afraid it won't be such a simple feature :).
I'll mark the issue as a feature request and leave it open.

richardriley999, 2024-07-07:
@FidelCapo -- thank you, I've been trying to figure how to do this, never thought to change the order in the right pane and could only get big slabs of alt color, maybe the feature request here is to update the UX to help people change the ordering (like up/down arrows for layers in image design apps) which will make it a bit more discoverable

closed 2026-09-29 by prusaslicer-bot[bot]