prusa3d/PrusaSlicer #11429: slic3rutils_tests: deadlock in the BoostThreadWorker

vpa1977, 2023-10-06:
### Description of the bug

slicr3rutils_tests hang indefinitely due to deadlock/race condition in BoostThreadWorker

### Project file & How to reproduce

Configured  and build the project:
```
cmake .. -DBUILD_TESTING=1 \
                -DCMAKE_BUILD_TYPE=Release \
                -DOPENVDB_FIND_MODULE_PATH=/usr/lib/x86_64-linux-gnu/cmake/OpenVDB \
                -DSLIC3R_FHS=1 \
                -DSLIC3R_WX_STABLE=1 \
                -DSLIC3R_GTK=3
make
```

ran the test
```
gdb --args ./slic3rutils_tests -s
run
```
Test was stuck for several minutes. 

```
home/bsp/slic3r-prusa/tests/slic3rutils/slic3r_jobs_tests.cpp:48: PASSED:
  REQUIRE( !worker.is_idle() )
with expansion:
  true

^C
Thread 1 "slic3rutils_tes" received signal SIGINT, Interrupt.
0x00007ffff3a94076 in __futex_abstimed_wait_common64 (private=0, cancel=true, 
    abstime=0x0, op=393, expected=0, futex_word=0x7fffffffd370)
    at ./nptl/futex-internal.c:57
57	./nptl/futex-internal.c: No such file or directory.
(gdb) info threads
  Id   Target Id                                          Frame 
* 1    Thread 0x7fffef6c7140 (LWP 4439) "slic3rutils_tes" 0x00007ffff3a94076 in __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, 
    expected=0, futex_word=0x7fffffffd370) at ./nptl/futex-internal.c:57
  2    Thread 0x7fffef6bb6c0 (LWP 4440) "slic3rutils_tes" 0x00007ffff3a94076 in __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, 
    expected=0, futex_word=0x5555571ee6f0) at ./nptl/futex-internal.c:57
  3    Thread 0x7fffeeeba6c0 (LWP 4441) "slic3rutils_tes" 0x00007ffff3a94076 in __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, 
    expected=0, futex_word=0x5555571ee6f0) at ./nptl/futex-internal.c:57
  4    Thread 0x7fffee6b96c0 (LWP 4442) "slic3rutils_tes" 0x00007ffff3a94076 in __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, 
    expected=0, futex_word=0x5555571ee6f0) at ./nptl/futex-internal.c:57
  5    Thread 0x7fffedeb86c0 (LWP 4443) "slic3rutils_tes" 0x00007ffff3a94076 in __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, 
    expected=0, futex_word=0x5555571ee6f0) at ./nptl/futex-internal.c:57
  11   Thread 0x7fffecdff6c0 (LWP 4449) "worker_thread"   0x00007ffff3a94076 in __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, 
    expected=0, futex_word=0x7fffffffd2c8) at ./nptl/futex-internal.c:57
(gdb) 
```

Deadlocked threads:
```
#0  0x00007ffff3a94076 in __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x7fffffffd2c8) at ./nptl/futex-internal.c:57
#1  __futex_abstimed_wait_common (cancel=true, private=0, abstime=0x0, clockid=0, expected=0, futex_word=0x7fffffffd2c8) at ./nptl/futex-internal.c:87
#2  __GI___futex_abstimed_wait_cancelable64
    (futex_word=futex_word@entry=0x7fffffffd2c8, expected=expected@entry=0, clockid=clockid@entry=0, abstime=abstime@entry=0x0, private=private@entry=0)
    at ./nptl/futex-internal.c:139
#3  0x00007ffff3a969fe in __pthread_cond_wait_common (abstime=0x0, clockid=0, mutex=0x7fffffffd278, cond=0x7fffffffd2a0) at ./nptl/pthread_cond_wait.c:503
#4  ___pthread_cond_wait (cond=0x7fffffffd2a0, mutex=0x7fffffffd278) at ./nptl/pthread_cond_wait.c:627
#5  0x00005555558f2b5b in Slic3r::GUI::BoostThreadWorker::run() ()
#6  0x00007ffff7ec9f3b in ??? () at /lib/x86_64-linux-gnu/libboost_thread.so.1.74.0
#7  0x00007ffff3a97ada in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:444
#8  0x00007ffff3b2845c in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:78
```

```
#0  0x00007ffff3a94076 in __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x7fffffffd370) at ./nptl/futex-internal.c:57
#1  __futex_abstimed_wait_common (cancel=true, private=0, abstime=0x0, clockid=0, expected=0, futex_word=0x7fffffffd370) at ./nptl/futex-internal.c:87
#2  __GI___futex_abstimed_wait_cancelable64
    (futex_word=futex_word@entry=0x7fffffffd370, expected=expected@entry=0, clockid=clockid@entry=0, abstime=abstime@entry=0x0, private=private@entry=0)
    at ./nptl/futex-internal.c:139
#3  0x00007ffff3a969fe in __pthread_cond_wait_common (abstime=0x0, clockid=0, mutex=0x7fffffffd320, cond=0x7fffffffd348) at ./nptl/pthread_cond_wait.c:503
#4  ___pthread_cond_wait (cond=0x7fffffffd348, mutex=0x7fffffffd320) at ./nptl/pthread_cond_wait.c:627
#5  0x00005555558f1be3 in Slic3r::GUI::BoostThreadWorker::wait_for_idle(unsigned int) ()
#6  0x00005555558d0dcf in void C_A_T_C_H_T_E_M_P_L_A_T_E_T_E_S_T_F_U_N_C_6<Slic3r::GUI::BoostThreadWorker>() ()
#7  0x0000555555870833 in Catch::RunContext::invokeActiveTestCase() ()
#8  0x00005555558895d8 in Catch::RunContext::runCurrentTest(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) ()
#9  0x00005555558922cb in Catch::RunContext::runTest(Catch::TestCase const&) ()
#10 0x000055555589c620 in Catch::Session::runInternal() ()
#11 0x000055555589cc0e in Catch::Session::run() ()
#12 0x00005555558391fb in main ()

```

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

Ubuntu 23.10

### Printer model

NA

lukasmatena, 2023-10-06:
Thanks, this is very useful information. We will investigate.

tamasmeszaros, 2023-10-09:
Hi @vpa1977 !
Could you please try if you still get the deadlock after applying the above commit (
a863381) ?

vpa1977, 2023-10-09:
Yes, this was detected during build time/autopkgtests which is a virtual infrastructure with high cpu contention and timeouts.

Would it be possible to consider using std::condition_variable[1] to resolve the issue?

[1] https://en.cppreference.com/w/cpp/thread/condition_variable

closed 2023-10-20 by lukasmatena