prusa3d/PrusaSlicer #11430: Bridging is starting in free air

freakydude, 2023-10-06:
### Description of the bug

Slice the famous pearl tower calibration model.

Bridges starting in free air (triangle in the screenshot, even the second line around is still starting in free air -> not connected)

[image]

The result looks not bad - but due the wrong order the filament will fall down

[image]

Playing with some slice parameters did not solve the problem

### Project file & How to reproduce

Open the project, look into the sliced timeline like in the screenshot above

[PolyPearl_Tower.zip](https://github.com/prusa3d/PrusaSlicer/files/12830635/PolyPearl_Tower.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1+win64

### Operating system

WIndows 11

### Printer model

Sidewinder X2 with Klipper

kdobrev, 2024-03-26:
I did not expect this bug to be so old. Is it something related to https://github.com/prusa3d/PrusaSlicer/issues/12440 that is also an old ticket, but not considered a BUG?

freakydude, 2024-03-27:
Update from my side: Issue still exists even in newest version 2.7.3-rc1

[image]

kdobrev, 2024-05-12:
PrusaSlicer is slowly but surely falling behind: https://github.com/prusa3d/PrusaSlicer/issues/12708

tastyratz, 2024-06-08:
This is most likely related to or really just a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/9074 persisting the last few years where Arachne starts bridges in thin air due to bridges printing out of order to all wall perimeters when more than 1 perimeter is selected

kubispe1, 2024-07-01:
This issue persists in PS 2.8.0 Final. The issue is relevant for both the Classic and Arachne generators, as both create similar issues.

freakydude, 2026-09-02:
Hi,

i retested the issue with the current PrusaSlicer 3.0.0-alpha11 and it's still the same

Started with default Prusa i3 MK3S Profile, Nozzle 0.2 and default Prusament PLA filament. 

- The issue does not exist with 2 Perimeters
- The issue exists with more than 2 Perimetes (e.g. 3 perimeters)

See Picture

[image]

best Regards
freakyDude