prusa3d/PrusaSlicer #11432: Organic supports start in mid air

bdwheele, 2023-10-06:
### Description of the bug

The organic supports under an overhanging part of a model are floating in mid air.

### Project file & How to reproduce

(file is not really zipped ... just added .zip extension)

[Dimetrodon.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/12833333/Dimetrodon.3mf.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1+linux-x64-GTK3

### Operating system

Fedora Linux 38

### Printer model

MK4 w/input shaper

FidelCapo, 2023-10-09:
I tried slice your project but I don't see any generated supports in the air..

[image]

FidelCapo, 2023-10-09:
Oh, I see. They're inside the other supports. I'll try to find out what's going on.

[image]

FidelCapo, 2023-10-09:
It's already reported here: https://github.com/prusa3d/PrusaSlicer/issues/9601 
I'll link the issues and close this as a duplicate. Thanks for the additional data.

closed 2023-10-09 by FidelCapo