prusa3d/PrusaSlicer #11433: DISASTROUS SEAMS (part 4 and continues...)

ramosglauco, 2023-10-06:
### Description of the bug

Another compilation of the TERRIBLE work that the seam algorithm has been doing since version 2.4.0. Here is every time I have reported this issue since June/2022 without any positive feedback: #8443 #9649 #11190

But I'm persistent and I won't stop reporting until they fix it. It's time to focus on the basics instead of other things.

NOW ALIGNED SEAMS ARE LIKE THIS:
[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

### Project file & How to reproduce

Use the cylinder from the shape gallery.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

Win11

### Printer model

FLSun, Artillery, BIQU and others

0gl0, 2023-10-08:
I want to add functionality：
[Seam gap](https://github.com/SoftFever/OrcaSlicer/wiki/Seam#seam-gap)
To address the issue of 'z-seam' gaps
[image]

ramosglauco, 2023-10-08:
They are more concerned with adding a bunch of crap new features than actually useful things. I bet in the next patch we will have more plug features! =) Meanwhile features like: only one top perimeter, really intelligent seam algorithm, elephant foot compensation decreasing on multiple layers, z-lift only on top infill, and tons of other good ideas, are forgotten in limbo.

murk-sy, 2023-10-10:
Please stop creating new issues, this isn't a forum. The point of an issue tracker is for reporting **distinct** issues or suggestions to improve things in specific ways. Ideally with a descriptive title.

It won't get it resolved any faster if you make a new issue every time you fancy it. You're supposed to pick one that already exists, ideally the earliest, or the one with the most information, and close the rest.

Yes, the seam algorithm is far from perfect, but it very much depends on the model itself. Seam painting does not always create nice lines either. It's an issue that annoys anyone using seam painting as well. I've had plenty of frustrations with it myself. Using very thin lines for seam painting is a fairly reliable workaround that works.

And just so we're clear, seams are a known issue for developers even without you reminding them every few months. Despite that, all of your existing issues have the seam label, meaning that the developers marked it and have added all of them to the todo list. Likely because they all contain examples, because that's what they're interested in.

@0gl0 Please make a separate feature request for that otherwise it'll just be lost to the ether.

bryn51, 2023-12-21:
I'll add my vote to this one, as I was about to report it myself.  Basically whether the seam line is aligned or random there is a gap between start of the seam line and the end or (in the case of Random) there are a bunch of tiny holes all over the surface of the print.  It's completely unacceptable, as those seam lines will leak Fluid or air, reduce the strength and detract from the visual appearance of the print.  
I do not accept that the gaps cannot be removed.  I'll be checking out "Classic" perimeter generator to see if it's any better.  This bug report certainly applies to the "Arachne" perimeter generator.

tabletopstor3d, 2024-03-23:
+1 would love to see this fixed. We recently jumped from 2.4 to 2.7 in order to support our XL and its been a huge step backwards.

Jan-Soustruznik, 2024-09-16:
closed as duplicate of https://github.com/prusa3d/PrusaSlicer/issues/8443

closed 2024-09-16 by Jan-Soustruznik