prusa3d/PrusaSlicer #11436: Bad infill in multiple parts

pecuhel, 2023-10-07:
### Description of the bug

I was creating a model that was supposed to display the types of fills.
But two fillings are wrong archimedean chord and octagram spiral.
It looks like the center of the model part takes the center of the whole print and not the center of this part.

### Project file & How to reproduce

[infil-test1-16.zip](https://github.com/prusa3d/PrusaSlicer/files/12837724/infil-test1-16.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

Windows 10

### Printer model

MK3s+

pecuhel, 2023-10-07:
[image]

neophyl, 2023-10-07:
Correct, that's how it works.  Always has. I've seen several existing requests to allow altering the infill 'centre; point over the years.
For example https://github.com/prusa3d/PrusaSlicer/issues/5482 which was one of the ones I remembered.

Jan-Soustruznik, 2023-10-09:
Issue closed as duplicate

closed 2023-10-09 by Jan-Soustruznik