prusa3d/PrusaSlicer #11438: Slicing stuck at 90%

viesturz, 2023-10-08:
### Description of the bug

The slicing of the following project gest stuck at 90%

### Project file & How to reproduce

Slice the attached project.

Note that there are two copies of the same item, slicing just one works fine, but slicing two never finishes.
[Spring.zip](https://github.com/prusa3d/PrusaSlicer/files/12840098/Spring.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

Win11

### Printer model

voron 2.4

Jan-Soustruznik, 2023-10-10:
Hello, @viesturz , 

I tried to slice your attached .3MF file ...  It take little bit more time on generate g-code(at 90%), but the project was sliced normally. - I used old lenovo laptop with 8GB ram, i5 cpu ..  Did you tried to take wait some time, if really slice stuck at 90% ... Could you share some info about your HW ... CPU/RAM/MB...

Thank you, have a nice day.

viesturz, 2023-10-10:
It eventually did slice, but had to wait > 30 seconds.
That is really long for such a simple part. Ryzen 7950 with 32Gb ram here.

viesturz, 2023-10-16:
> Hello, @viesturz <https://github.com/viesturz> ,
>
> I checked your project file more deeply. I found that you have an enabled
> feature - *Avoid crossing perimeters*, that makes sense now why the
> generate time of G-code is slower with more models on bed.
>
> from help(link:
> https://help.prusa3d.com/article/layers-and-perimeters_1748#avoid-crossing-perimeters
> ):
>
> Enabling this feature slows down the G-code generation and increases print
> time somewhat (time increase varies between models).

Sorry but that is a non answer. I have been using that setting without
issue, slicing tens of objects with that setting in seconds.

Will check back later once I get to my PC,  could be that avoid crossing
perimeters is crazy slow with spirals.

PaparazziN, 2023-10-21:
Hi @viesturz!
As you mentioned previously the file does slice but it takes a while. The crossing perimeters does seem to act up a bit on that model, which I assume is due to the long, thin and spiraling perimeters. I agree that in most cases using 'avoid crossing perimeters' is barely noticeable when slicing but in this case the opposite seems to be true, turning it off slices the model almost immediately on my PC. Just for fun I also tried changing 'Max detour length' to zero, compared to 100 mm (as you had it), resulted in it slicing almost twice as fast on my PC, so that also seems to play a role YMMW though.

I would assume that these types off edge cases is the reason that 'avoid crossing perimeters' is off by default. Whilst you have absolutely found some improvements that can be made with the 'avoid crossing perimeters' function, I would consider the original issue you mentioned as closed. If you like I would suggest you make a new issue that requests for improvements with the 'avoid crossing perimeters' function.

viesturz, 2023-10-21:
Sounds good, I will open a new ticket then.

closed 2023-10-21 by viesturz