prusa3d/PrusaSlicer #11440: Organic support infill

LoneSimba, 2023-10-08:
**Is your feature request related to a problem? Please describe.**
RN organic supports generate single\double wall support with no infill. It works for regular nozzle sizes (e.g. 0.4, 0.6, 0.3), but for 0.1 even double wall is just too thin - ot has to be at least 4 to be similar to 0.4 single wall. Add insult to injury, w\o infill it will eventually break due to heat from the nozzle while making the required amount of walls

**Describe the solution you'd like**
An option to allow infills in supports, as well as better control over support parameters (amount of wall and such)

**Describe how it would work**
As this is likely rare case, enable these options only for advanced\expert modes, for organic support add a checkbox "enable infill", when checked - enable infill props, else - disable them (rn infill props are always enabled, even tho organic has no infill)

**Describe alternatives you've considered**
Sticking with grid + honeycomb infill

xxTHECHADxx, 2025-02-05:
New to this site. Not sure if this is the location to post this. 

I'd love to see infill for organic supports. I run a large printer and print some tall things that need tall/long organic supports that could benefit from some infill support. I used to slice with Cura and used that function often. Love Prusa slicer, keep up the great work.

rcmaniac25, 2026-09-13:
More options to ensure stability of supports would be nice to have.