prusa3d/PrusaSlicer #11445: Slicer crashes while making infill

Swarozyc, 2023-10-10:
### Description of the bug

While slicing charybdis keyboard, slicer crashes during "Making infill" part. Without any error, without notification, app closes.

### Project file & How to reproduce

[charybdisnano_v2_v187.zip](https://github.com/prusa3d/PrusaSlicer/files/12857038/charybdisnano_v2_v187.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1+win64

### Operating system

Windows 11

### Printer model

Voron Trident

Lagersuufer, 2023-10-11:
I am able to recreate the same issue. I have a multi-material project that was created for the MK3s+/MMU2s. Slices just fine. When I try to convert this to the XL 5 head, I load the project, then change to the XL and migrate all the current painting settings and filaments. When I trigger the slicing, it gets to "Making infill" and crashes to desktop.  Windows 11, PrusaSlicer 2.6.1.

lukasmatena, 2023-10-11:
Thanks. This has the same cause as #11267, which is already fixed. The fix will be in the next release. Closing.

closed 2023-10-11 by lukasmatena