prusa3d/PrusaSlicer #11447: bande de couleurs en mode vase

18pool49, 2023-10-11:
### Description of the bug

une pièce faite en mode classique parois int. + remplissage + parois ext. 
est bien réalisé en bande de couleur a la demande sans problème >>> 
[image]
mais si je recommence cette même pièce en cochant >> 
[image]
et que je désire les bandes que je définies >> je découpe  !!!!!!!    et ca ne fonctionne PAS  
[image]

### Project file & How to reproduce

[martiniglass3 projet vase dd.zip](https://github.com/prusa3d/PrusaSlicer/files/12866249/martiniglass3.projet.vase.dd.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

W11

### Printer model

TLD3 PRO Tenlog

neophyl, 2023-10-11:
That doesn't look like a valid model to be printed in vase mode.  Vase mode is only possible when it can print using a continuous extrusion, this generally means you model a SOLID volume and vase mode then prints the exterior and no top.  The model you present has an interior volume and so isn't valid.

18pool49, 2023-10-11:
bonjour et merci pour votre rapide réponse 
mais vous me surprenez car c'est pas la 1er fois que je prend un volume pour en faire un vase sur Prusa dans des versions antérieur  
je viens de prendre le même volume sur Ideamaker que j'ai coupé a différentes hauteur , et je demande un vase en Bi couleur et il me le sort correctement  
[image]
[martiniglass3 projetdd tlid.zip](https://github.com/prusa3d/PrusaSlicer/files/12869655/martiniglass3.projetdd.tlid.zip)
voilà le Gcode (Zip) 
je vais chercher dans mon historique de Gcode

18pool49, 2023-10-11:
je viens d'en retrouver un vase en partant d'un volume  avec prise d'écran de 2022 sur prusa 
les bandes sont faite sur Prusa 
[image]
je vous donne le 3mf et le gcode en zip 
[bowl1.zip](https://github.com/prusa3d/PrusaSlicer/files/12869754/bowl1.zip)
[bowl1 bi b0.8 hc0.25 lc0.8.zip](https://github.com/prusa3d/PrusaSlicer/files/12869756/bowl1.bi.b0.8.hc0.25.lc0.8.zip)

je vais refaire le même Gcode de ce vase avec bande sur la version actuelle de prusa (2.6.1) et je vous tiens au courant

closed 2026-09-29 by prusaslicer-bot[bot]