prusa3d/PrusaSlicer #11449: "Emboss" tool to retain settings between uses

nrnrnr, 2023-10-11:
I'm always frustrated when I have a whole bunch of text to add using "add part::text" on a face, because for every single text item I have to reset the font size, depth, and join/cut radiobutton.  The solution I would like would be for these settings _not_ to revert to their defaults between uses, but to remain unchanged.  Ideally the settings would persist between PrusaSlicer sessions, but it would be sufficient for them not to be reset within a single session.

An alternative I've considered would be to have a way to specify the defaults within a configuration file.  Such a feature would also be nice, but the value for effort would be lower than simply retaining settings.

The version I am using is Version 2.6.1+linux-x64-GTK3.

Jony01, 2023-10-24:
Exactly for this purpose is there the styles.
[image]
Style keeps settings of text as you mentioned: font size, depth, ...

closed 2023-10-24 by Jony01

nrnrnr, 2023-10-24:
Brilliant!  I shall look forward to figuring that out.