prusa3d/PrusaSlicer #11456: Error encountered when opening gcode generated by PS in PS GCode Viewer

coldyoungguy, 2023-10-12:
### Description of the bug

Currently using Prusaslicer 2.6.1 and G-code Viewer 2.6.1

During my print of the gcode below using octoprint, the print did not actually complete, however, in Octoprint, it is stated to be completed. This made me suspect that maybe the uplaod of the gcode file via the octoprint API may have messed with the gcode.

However, upon reslicing a brand new file via PS and checking it in the g-code viewer, i get the error below.
[image]

There does not seem to be any errors like it from a quick google search.

### Project file & How to reproduce

[BB4 (2).zip](https://github.com/prusa3d/PrusaSlicer/files/12885324/BB4.2.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.1+win64-202309060711

### Operating system

Windows 11 (build 22621), 64-bit edition

### Printer model

Prusa i3 MK3S

lukasmatena, 2023-10-13:
@coldyoungguy I cannot open the 3MF, it is corrupted. There is some garbage in "compatible_printers_condition_cummulative" config option. Were you doing some non-standard manipulations with the profiles before the problem occurred?

coldyoungguy, 2023-10-18:
@lukasmatena Thank you for your response, upon reading that you have found some random stuff in the configs, i went to check each profile and it seems that the profile dependencies was cut off for some reason. It be very much due to my own doing a while back. Upon restoring the dependencies to the default ones, i can now open the gcode in the gcode viewer. Thanks for your help.