prusa3d/PrusaSlicer #11458: PrusaSlicer crashes when using the configuration wizard filaments tab

kevinbiehl, 2023-10-12:
### Description of the bug

The configuration wizard encounters an error when accessing the Filaments tab and selecting the PETG profile from "Das Filament". Instead of importing the profile, the slicer crashes and closes without any error message.

Profile Selection Setup:
[image]

### Project file & How to reproduce

1. Open configuration wizard (Top Bar: Configuration > Configuration Wizard...).
2. Click straight on the "Filaments" tap.
3. Select Printer: (All); Type: PETG; Vendor: Das Filament
4. Check box on Profile "Das Filament PETG"
5. Press the bottom Finish button

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.1+win64

### Operating system

Windows 10 

### Printer model

Ender 3 Pro with Marlin

neophyl, 2023-10-13:
Probably the same issue as https://github.com/prusa3d/PrusaSlicer/issues/11390

rtyr, 2023-10-13:
Indeed. We believe we managed to fix it for next release.

closed 2023-10-13 by rtyr