prusa3d/PrusaSlicer #11460: Holowing creates a highly tesselated duplicate outer shell geometry causing non-manifold issues with other software

The5-1, 2023-10-13:
### Description of the bug

Since a while (I unfortunately don't know when) but currently 2.6.1 the hollow algorithm generates a new mesh for the outer shell.
Every triangle is densely tessellated and this densely tessellated mesh is merged into the existing base mesh.

I assume a tessellated version of the outer shell, that is used for the hollowing algorithm, is generated during the process but not "cleaned up" by the end of it anymore.

Here you see a big original triangle and the tessellated triangles at the same location in the mesh, observe the distinct z-fighting:
[image]

[image]

Unfortunately this causes issues with processing the STL any further (such as other manufacturer specific Resin printer slicers) as the geometry is now non-manifold.

I saw #11053 but Made this for the sake of more information:

### Project file & How to reproduce

Load any STL, got to a SLA printer, hollow the model with any parameters.
Export STL and inspect the generated geometry e.g. in bender.
Most viewers will show "flickering" faces due to duplicate face geometry.
Pluck the geometry apart and observe it being 2 identical meshes merged, one densely tessellated and the original geometry.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

Windows 10

### Printer model

no specific, the meshes are non-manifold

The5-1, 2024-02-24:
This is still present in 2.7.1

jpjokela, 2024-03-08:
I can confirm this happening also on macOS 14.2.1 in PrusaSlicer 2.7.2 on M1 machine.

Just downgraded to 2.5.2 as this is pretty much must-have feature to me. Tried a few versions in-between, and they seemed to break the mesh in similar ways...

(Guess I'll stay on this version quite a while...)

jpjokela, 2024-03-08:
I can confirm this happening also on macOS 14.2.1 in PrusaSlicer 2.7.2 on M1 machine.

Just downgraded to 2.5.2 as this is pretty much must-have feature to me. Tried a few versions in-between, and they seemed to break the mesh in similar ways...

(Guess I'll stay on this version quite a while...)

-EDIT-

Of 6 models that I hollowed recently with PrusaSlicer 2.7.2, for some reason, one ended up having valid mesh. No idea why that one worked. Just wanted to point out that failure rate isn't 100%, if that helps with finding the bug (still way too high to be usable)

jpjokela, 2024-07-21:
Still broken on 2.8.0 on ARM mac.

I wonder if anyone will look into this, unless you upload a "project file"? (issue shows 1 of 2 tasks complete) - I guess I can't do it.

If it helps, here's a free model that reproduces the issue;
https://www.myminifactory.com/object/3d-print-kaarija-bust-pre-supported-301419
Inside you will find TTM_Kaarija/TTM_KaarijaBust.stl

If you import that to latest PrusaSlicer, hollow it with defaults without adding drain holes and export it, the model will be seriously broken. LycheeSlicer will do seemingly random shapes in most layers.

NOTE: Hollowing this particular model with defaults ends up with several closed resin pockets. Using thinner wall thickness might fix that, but decided to use defaults to simplify instructions, as it highlights the issue nicely anyway.

This shows the first layer of the previously mentioned hollowed model sliced with LycheeSlicer, which is obviously very broken (Of course this is also issue with bad error tolerance in LycheeSlicer, since Chitubox & PrusaSlicer print the model just fine. I still wish PrusaSlicer would create valid geometry when hollowing, like v2.5.2 did)
[image]

jpjokela, 2026-09-19:
While it feels unlikely that this will ever get fixed (and I have to stay at version 2.5.2 forever), keeping this issue alive, since my previous post has easy instructions for reproducing the issue.