prusa3d/PrusaSlicer #11462: Better support for multitool printers (like Prusa XL)

neslekkim, 2023-10-13:
Printing multiple materialtypes in same prints have some challenges, for instance PLA and TPU, check this for reference:
https://www.youtube.com/watch?v=oUXeIEUS68I

Points from the video:

You can't choose different speed for various head (TPU requires lower speed than PLA), and here he shows that he tweaks volumentric speed to make this work, instead of having this option pr tool.

Multiple nozzle sizes at same time cannot be used at same print, if you want one part of the print with 0.4mm, and other part with 0.6mm nozzle.

taxilian, 2023-11-27:
I'm really hoping we see work on this soon as the XL gets out there =] I have mine and it's fantastic but some of the benefit I hoped for is lost because I can't mix 0.6mm and 0.25mm nozzles (for detail).

Even if there was a restriction that it had to be a single nozzle size per layer there is a lot that could be done with it...

KlausKraemer, 2024-02-18:
Still no progress in supporting different nozzle sizes for the Prusa XL? That's quite disappointing, coming from the Ultimaker 3 and Cura, where this didn't pose any problem!

KlausKraemer, 2024-07-14:
> Try orcaslicer, it works: https://www.youtube.com/watch?v=4aFHm5xjPYs

That's not, what I expect or want to do. If PRUSA sells a mutlitool 3D printer, then I expect Prusa Slicer using the whole spectrum of possibilities.

m10d, 2024-07-19:
> Try orcaslicer, it works: https://www.youtube.com/watch?v=4aFHm5xjPYs
> 
> [SoftFever/OrcaSlicer#6087](https://github.com/SoftFever/OrcaSlicer/pull/6087)

No offense, but this statement isn't that helpful / somewhat inaccurate - at least not until there is a lot more clarity on that thread about how robust the solution is & a (possible) timeline to get it upstream. There is enough going on there that it's not clear how well (not saying it _does not_, just saying it's not established) the changeset meets the needs for the feature in general, or what other limitations it comes with. Also, whatever video you linked is not helpful in resolving the above.  It's > 10min long, and yes I can see it printing, yes there look to be some wipe towers - but it's not in any way a document that communicates the progress of another product into this repo.

That said, appears to be some fantastic progress on that work in Ocra, I am also very eager to be able to easily (at least _somewhat_ easily) mix nozzle sizes in reasonable ways.   

What I came looking for was more what the prusa team has to say about where this is on the roadmap - merging real designs with disparate material is complicated enough, I am sure we can all agree the end goal is official/buttoned up support for this - **do we have any clue what official maintainers think of feasibility (& priority) of addressing this?**

Many thank for everyone's work - just getting started on my XL.  This feature is significant as it unlocks a massive time-savings for large parts while still being able to keep small features toleranced and text looking professional.

patate1684, 2024-09-25:
Any news on this?
Will this be added to the roadmap ?

lzengineering, 2024-12-13:
After I recently changed one nozzle on my Prusa XL 2-tool to 0.5 mm diameter, I was also very disappointed by this limitation with the prime tower.
But broken down what a different nozzle diameter means is, in my opinion, the extrusion width. 
Therefore, a possible workaround is to pretend the printer that both extruders have the same nozzle diameter and to assign a special extrusion width to the part with the different diameter. This can be done in PrusaSlicer by right-clicking on part - Add setting - Extrusion width. 

Depending on how you have configured this in the profile, you have to adjust one or more values.
This can also be combined if different extruders are used for infill and perimeter.

In my first test print with 0.4 mm on extruder 1 and 0.5mm on extruder 2, this worked well.

If you have set the nozzle diameter to the real value in the printer-ui, the printer warns you that the wrong nozzle is selected. 

I would also like to see PrusaSlicer adapted soon so that this workaround is not necessary.

closed 2026-09-29 by prusaslicer-bot[bot]