prusa3d/PrusaSlicer #11463: Bug in Arachne - gap in thin walls

Eiim, 2023-10-13:
### Description of the bug

When slicing this model under Arachne, the path produced is missing part of a thin wall. With the classic generator, it fills the section of the wall correctly.

Model:
[image]

Classic perimeter generator, detect thin walls:
[image]

Classic perimeter generator, don't detect thin walls:
[image]

Arachne perimeter generator:
[image]

### Project file & How to reproduce

[PinTest.zip](https://github.com/prusa3d/PrusaSlicer/files/12897549/PinTest.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.1+win64

### Operating system

Windows 10

### Printer model

Prusa MINI

Eiim, 2023-10-13:
It happens at different detail levels. At .05mm, one corner gets noticeably worse:
[image]

At .25mm, the layers are about as large as the difference, but still on the bottom it appears:
[image]

neophyl, 2023-10-13:
The standard settings for arachne are ok for a lot of things but they are NOT a one set fits all.

You can alter the settings to get better results on models.  For example just quickly changing 2 settings on your model gives me this
[image]

I changed Perimeter transitioning threshold angle and minimum feature size.  Im sure if I adjusted further an even better result could be obtained.
[image]

In fact changing minimum feature size to 100% gives this on that bottom area
[image]

closed 2026-09-29 by prusaslicer-bot[bot]