prusa3d/PrusaSlicer #11465: Slicing a specific model completely crashed PrusaSlicer

mpiotrowski, 2023-10-13:
### Description of the bug

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               PrusaSlicer [1384]
Path:                  /Applications/Original Prusa Drivers/PrusaSlicer.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.6.1+MacOS-x64-202309060812 (PrusaSlicer-2.6.1+MacOS-x64-202309060812)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-10-13 18:33:41.4647 -0400
OS Version:            macOS 14.0 (23A344)
Report Version:        12
Anonymous UUID:        F820056C-8DE3-B622-05F5-FEB01EBF112D

Time Awake Since Boot: 150 seconds

System Integrity Protection: enabled

Crashed Thread:        11  slic3r_tbb_3

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [1384]

VM Region Info: 0 is not in any region.  Bytes before following region: 4328669184
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      102024000-104fe0000    [ 47.7M] r-x/r-x SM=COW  ...S/PrusaSlicer

Thread 0:: slic3r_main Dispatch queue: com.apple.main-thread
0   Metal                         	       0x1923c0cd8 -[_MTLCommandBuffer commandQueue] + 0
1   AGXMetalG14X                  	       0x1f1bbb110 -[AGXG14XFamilyCommandBuffer initWithQueue:retainedReferences:] + 84
2   AGXMetalG14X                  	       0x1f1bbd4e4 -[AGXG14XFamilyCommandQueue commandBufferWithUnretainedReferences] + 44
3   AppleMetalOpenGLRenderer      	       0x1135f6acc GLDContextRec::getPreprocessingCommandBuffer() + 56
4   AppleMetalOpenGLRenderer      	       0x1135f6b48 GLDContextRec::startPreprocessingComputeEncoderIfNeeded() + 88
5   AppleMetalOpenGLRenderer      	       0x113603ae4 gldRenderVertexArray(GLDContextRec*, unsigned int, unsigned int, int, int, unsigned int, void const*, int, void const*) + 7184
6   GLEngine                      	       0x1f0071b50 glDrawElements_GL3Exec + 288
7   PrusaSlicer                   	       0x102a22034 Slic3r::GUI::GLModel::render(std::__1::pair<unsigned long, unsigned long> const&) + 1000
8   PrusaSlicer                   	       0x102a21c40 Slic3r::GUI::GLModel::render() + 44
9   PrusaSlicer                   	       0x102a2b820 Slic3r::GUI::GLTexture::render_sub_texture(unsigned int, float, float, float, float, Slic3r::GUI::GLTexture::Quad_UVs const&) + 896
10  PrusaSlicer                   	       0x102a30ff4 Slic3r::GUI::GLToolbar::render_background(float, float, float, float, float, float) const + 636
11  PrusaSlicer                   	       0x102a2f9a4 Slic3r::GUI::GLToolbar::render_horizontal(Slic3r::GUI::GLCanvas3D const&) + 292
12  PrusaSlicer                   	       0x102937d44 Slic3r::GUI::GLCanvas3D::_render_view_toolbar() const + 204
13  PrusaSlicer                   	       0x1029206c4 Slic3r::GUI::GLCanvas3D::_render_overlays() + 168
14  PrusaSlicer                   	       0x10291d2c8 Slic3r::GUI::GLCanvas3D::render() + 1184
15  PrusaSlicer                   	       0x10292c6d4 Slic3r::GUI::GLCanvas3D::_refresh_if_shown_on_screen() + 140
16  PrusaSlicer                   	       0x102928790 Slic3r::GUI::GLCanvas3D::on_idle(wxIdleEvent&) + 324
17  PrusaSlicer                   	       0x104727b80 wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 252
18  PrusaSlicer                   	       0x10472784c wxEvtHandler::ProcessEventLocally(wxEvent&) + 88
19  PrusaSlicer                   	       0x104727674 wxEvtHandler::ProcessEvent(wxEvent&) + 56
20  PrusaSlicer                   	       0x104727d2c wxEvtHandler::SafelyProcessEvent(wxEvent&) + 24
21  PrusaSlicer                   	       0x104580890 wxWindowBase::SendIdleEvents(wxIdleEvent&) + 92
22  PrusaSlicer                   	       0x1045808b8 wxWindowBase::SendIdleEvents(wxIdleEvent&) + 132
23  PrusaSlicer                   	       0x1045808b8 wxWindowBase::SendIdleEvents(wxIdleEvent&) + 132
24  PrusaSlicer                   	       0x1045808b8 wxWindowBase::SendIdleEvents(wxIdleEvent&) + 132
25  PrusaSlicer                   	       0x1045808b8 wxWindowBase::SendIdleEvents(wxIdleEvent&) + 132
26  PrusaSlicer                   	       0x1044c2784 wxAppBase::ProcessIdle() + 148
27  PrusaSlicer                   	       0x10461ba3c wxApp::ProcessIdle() + 36
28  PrusaSlicer                   	       0x104730508 wxCFEventLoop::OSXCommonModeObserverCallBack(__CFRunLoopObserver*, int, void*) + 124
29  CoreFoundation                	       0x1882330cc __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 36
30  CoreFoundation                	       0x188232fb8 __CFRunLoopDoObservers + 532
[... 246 lines cut ...]
15  PrusaSlicer                   	       0x1028f6c7c boost::detail::thread_data<Slic3r::BackgroundSlicingProcess::start()::$_2>::run() + 20
16  PrusaSlicer                   	       0x102e85ae0 boost::(anonymous namespace)::thread_proxy(void*) + 176
17  libsystem_pthread.dylib       	       0x188157034 _pthread_start + 136
18  libsystem_pthread.dylib       	       0x188151e3c thread_start + 8

### Project file & How to reproduce

Opening the 3mf file crashes PrusaSlicer [Cable holding wheel.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/12899873/Cable.holding.wheel.3mf.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1 MacOs

### Operating system

MacOS Sonoma 14.0

### Printer model

MINI+ but doesn't matter as I can't slice at all

JohnOCFII, 2023-10-14:
As an additional data point, I was successfully able to successfully open and slice the file:

```

Version 2.6.1+MacOS-arm64 - macOS Ventura 13.6 on a MacBook Pro M1/arm
```

[image]
[image]

Jan-Soustruznik, 2025-04-02:
Hello @mpiotrowski, Could you try the latest PS release, version 2.9.1? It seems that the crash has already been fixed. Please let us know. Thank you, and have a great day!

mpiotrowski, 2025-08-26:
Yes I haven't had this issue again, thank you!

closed 2025-08-26 by mpiotrowski